lihuang3/ur5_ROS-Gazebo #3: How to use this in a real  ur robot?

WwYyFan, 2018-09-12:
I am new to ROS,, if I want to run in a real ur robot,how can I start?

Abduoit, 2018-09-12:
@WwYyFan 

Did you run it with gazebo first,  I run the following without errors but the arm does not move !!

```
roslaunch ur5_notebook initialize.launch

python testvisionplanner.py
```

lihuang3, 2018-09-12:
@WwYyFan Please refer to https://github.com/lihuang3/ur3_ROS-hardware.git for hardware implementations. However, the hardware programs are based on UR3 instead of UR5. We are going to polish our ROS repos after ICRA 2019 submissions.

lihuang3, 2018-09-12:
@Abduoit I suggust use to try the following code first, https://github.com/lihuang3/ur5_ROS-Gazebo/blob/master/src/motionplanner2.py. Sometimes Cartesian path planning might not work because of the joint limit settings in UR configurations. The default [-pi, pi] joint range may not work for some planned goals.  
We are going to polish our ROS repos after ICRA 2019 submissions.

Abduoit, 2018-09-12:
@lihuang3 
I already run this, it is working fine. I was thinking to run the tracking part with red box, to understand more the code. I am interested more with vision tracking and kinematics.

WwYyFan, 2018-09-13:
@Abduoit @lihuang3 I have run it with gazebo first,the arm can move , but can't  visually track red boxes .
```
roslaunch ur5_notebook initialize.launch

python testvisionplanner.py
```

lihuang3, 2018-09-13:
@WwYyFan Only red box detection and motion planning modules are tested in this .py file. To follow/track the  red box, please refer to [`ur5_vision.py`] (https://github.com/lihuang3/ur5_ROS-Gazebo/blob/master/ur5_vision.py) and [`ur5_mp.py`](https://github.com/lihuang3/ur5_ROS-Gazebo/blob/master/ur5_mp.py)

WwYyFan, 2018-09-13:
@lihuang3 Thanks for your reply. I have try  ur5_vision.py, but the arm does not move and terminal does not have any output , when I try ur5_mp.py,the terminal output 
[image]

how to fix it ?

lihuang3, 2018-09-13:
@WwYyFan Because those are not standalone files. The proper way to run it is using roslaunch command:
```
roslaunch ur5_notebook initialize.launch
```

WwYyFan, 2018-09-13:
@lihuang3  I have roslaunch ur5_notebook initialize.launch before python ur5_mp.py

lihuang3, 2018-09-13:
@WwYyFan That's weird. Does Gazebo display the conveyor and moving red boxes?

Haoran-Zhao, 2018-09-13:
And did you see the camera view window?

WwYyFan, 2018-09-13:
@lihuang3  yes, Gazebo display the conveyor and moving red boxes.
[image]

WwYyFan, 2018-09-13:
@Haoran-Zhao  I can't see the camera view window

lihuang3, 2018-09-13:
@WwYyFan When you use roslaunch, all related nodes for this demo would be intialized. Soyou don't need to run .py files separately. Please check the camera module in URDF or related files. It seems the camera is not configured.

WwYyFan, 2018-09-13:
@lihuang3 I check the conveyor_belt.sdf , Can't find anything about the camera module,Is the code you uploaded incorrect?

lihuang3, 2018-09-13:
@WwYyFan It should be within these two files: 
```
      ur5_ROS-Gazebo/src/ur_description/ur5.urdf.xacro
      ur5_ROS-Gazebo/src/ur_description/common.gazebo.xacro
```    
You can either replace the default files in `UR5/ur_description` with these files, or grab the camera module config and insert it into the default files.

WwYyFan, 2018-09-13:
@lihuang3  I have added the file to that location , but still can't not see the camera .

WwYyFan, 2018-09-13:
@lihuang3  I check the <a href="https://github.com/lihuang3/ur5_ROS-Gazebo/launch/initialize.launch"><code>initialize.launch</code></a> again and  can't not find  where to load <a href="https://github.com/lihuang3/ur5_ROS-Gazebo/src/ur_description/ur5_joint_limited_robot.urdf.xacro"><code>ur5_joint_limited_robot.urdf.xacro</code></a>

Haoran-Zhao, 2018-09-13:
> @lihuang3 <https://github.com/lihuang3> I check the initialize.launch
> again and can't not find where to load ur5_joint_limited_robot.urdf.xacro

I dont have my computer right, but if you look at the launch file there is
a section for spawning ur5 model.
It’s in another folder for universal robot. Your will find them under ur5
folder.
You can try to search ur_description in the workspace folder.

lihuang3, 2018-09-13:
@WwYyFan The Universal Robot itself should call .xacro file, which part of UR, not initialize.launch. 
Please refer to the offical launch file: [`ur5_upload.launch`](https://github.com/ros-industrial/universal_robot/blob/kinetic-devel/ur_description/launch/ur5_upload.launch)
and the default .xacro files [`common.gazebo.xacro`](https://github.com/ros-industrial/universal_robot/blob/kinetic-devel/ur_description/urdf/common.gazebo.xacro), [`ur5.urdf.xacro`](https://github.com/ros-industrial/universal_robot/blob/kinetic-devel/ur_description/urdf/ur5.urdf.xacro)

WwYyFan, 2018-09-13:
@lihuang3 Thanks a lot ,  I will wait until next week.

WwYyFan, 2018-09-13:
Looking forward to your update

Abduoit, 2018-09-13:
I am getting same as @WwYyFan 
the conveyor is moving with red boxes, but there is camera mounted on the arm.

WwYyFan, 2018-09-18:
@lihuang3 Thanks for your update, I have done it with the new README.md , I can see the camera module and vacuum gripper module now ,but the robotic arm can only pick up the red box and cannot put it in the bule box. and the terminal output error message:

```
[ERROR] [1537234590.091685889, 28.220000000]: Trajectory message contains waypoints that are not strictly increasing in time.
[ WARN] [1537234590.091792380, 28.220000000]: Controller  failed with error code INVALID_GOAL
[ WARN] [1537234590.091825310, 28.220000000]: Controller handle  reports status FAILED
[ INFO] [1537234590.091851421, 28.220000000]: Completed trajectory execution with status FAILED ...
[ INFO] [1537234590.091920918, 28.220000000]: Execution completed: FAILED
gripper status = True
[ERROR] [1537234590.092339, 28.221000]: bad callback: <bound method ur5_mp.tracking_callback of <__main__.ur5_mp instance at 0x7f091931f878>>
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/efun/git/learn/ur_learn/src/ur5_ROS-Gazebo/ur5_mp.py", line 174, in tracking_callback
    self.execute()
  File "/home/efun/git/learn/ur_learn/src/ur5_ROS-Gazebo/ur5_mp.py", line 236, in execute
    self.arm.set_joint_value_target(self.transition_pose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/moveit_commander/move_group.py", line 227, in set_joint_value_target
    raise MoveItCommanderException("Error setting joint target. Is the target within bounds?")
MoveItCommanderException: Error setting joint target. Is the target within bounds?
```

lihuang3, 2018-09-18:
@WwYyFan 
According to 
```
 line 227, in set_joint_value_target
    raise MoveItCommanderException("Error setting joint target. Is the target within bounds?")
MoveItCommanderException: Error setting joint target. Is the target within bounds?
```
Did you change the joint limit as mentioned in the tutial?
```
  shoulder_pan_lower_limit="${-2*pi}" shoulder_pan_upper_limit="${2*pi}"
```
Or you can try to set all joint limit to [-2*pi, 2*pi] in`ur5_joint_limited_robot.urdf.xacro`

WwYyFan, 2018-09-18:
@lihuang3 Thanks a lot , I modify the ur5_joint_limited_robot.urdf.xacro , and it's work now.I will close this issue, thank you for your guidance during this time.

closed 2018-09-18 by WwYyFan

lihuang3, 2018-09-18:
@WwYyFan Sounds great!

WwYyFan, 2018-09-18:
@lihuang3 I will try  [`ur3_ROS-hardware`](https://github.com/lihuang3/ur3_ROS-hardware.git)  next.If I encounter a problem, I will ask you again.Thank you or your help !