lihuang3/ur5_ROS-Gazebo #4: Gazebo vacuum gripper plugin

st99315, 2018-09-19:
Hi! I have seen your video of pick and place demo using UR5 and vacuum gripper from youtube.
This is a nice work and we are very interesting!!

Recently we want to use vacuum gripper in gazebo simulator, but we get some problem. 

We can load plugin to simulator and trigger it, but  sometime vacuum gripper can’t grasp object in gazebo.

We’d like to ask you some questions:
1. What reason make you increase amount of vacuum gripper becoming nine grippers?
2. What kind of help would do this?

Thanks for providing useful example on github!

lihuang3, 2018-09-19:
@st99315 Thanks for your interest!

1. & 2.  It seems that each vacuum gripper only provides limited force to lift object. We spent two or three days investigating the vacuum gripper module, but did not figure out an elegant way to handle even a very light box. So we just use 9 of them to make sure we have enough force and meanwhile reduce the weight of objects.
3.  Looks like these are the effective range for the vacuum gripper.  I'm not sure, but I have to say, the vacuum gripper module is not well described, and they did not behave as well as what they meant to.

I suggest you refer to the following websites, as they discussed some interesting topics about the module:
https://answers.ros.org/question/243080/need-help-to-trigger-the-vacuum-gripper/
https://github.com/ros-simulation/gazebo_ros_pkgs/pull/389
http://answers.gazebosim.org/question/6306/how-to-simulate-vacuum-clamping/

bharatm11, 2018-12-04:
Hi @lihuang3 
Do you think just adding the vacuum grippers to the URDF and calling the appropriate service from the terminal will do the job? What is the role of ur5_gripper.py ?
Thanks,
Bharat Mathur

bharatm11, 2018-12-04:
@lihuang3 Thank you very much for your reply! 
I tried using your URDF to spawn the UR5 on Gazebo and called the services to turn the gripper on using the terminal but it fails to pick the object up. It works when I try to run your launch file.
I even made sure that the gripper feedback topics /ur5/vacuum_gripper/grasping were true when I called the service.
What else do you think I should try?

lihuang3, 2018-12-04:
@bharatm11 How many grippers did you turn on? One or two probably won’t make it happen, as the vacuum force is very limited. You might need about 5 grippers for a red box in this project. Also please make sure that the grippers have fully contact with the target object surface.

lihuang3, 2018-12-05:
@bharatm11 Sounds great!

garethByers, 2021-09-30:
I have used this gripper and have added it to the ur5_robot however, it still wont grasp when they are set  to on. i can look in the topic list and they are there and are set to on in the gazebo terminal but they still do not move the object. Is there any advice you have?