lihuang3/ur5_ROS-Gazebo #5: How can I get the same goal as gif

weiwuhuhu, 2018-09-28:
Hello Lihuang：
      I had followed your steps and installed all the package which in the readme, but how can I get the same effect after I had executed the step of "roslaunch ur5_notebook initialize.launch".But the ur5 remained static, it can't grasp the red box.

weiwuhuhu, 2018-09-28:
I got it, one step i missed.Thank you LiHuang, you are so cool!!!

aositeluofu, 2019-03-14:
Hi,I would like to know how to let the robot arm move to grasp the red box.I am still new to this ROS area.Can you tell me what launch file should I execute? looking forward to your reply

aositeluofu, 2019-03-14:
@weiwuhuhu Hi,I would like to know how to let the robot arm move to grasp the red box.I am still new to this ROS area.Can you tell me what launch file should I execute? looking forward to your reply

learningxiaobai, 2021-04-23:
> I got it, one step i missed.Thank you LiHuang, you are so cool!!!

hello，which steps you missed？my robot is static