lihuang3/ur5_ROS-Gazebo #6: What is mean of this code?

weiwuhuhu, 2018-09-29:
Hello LiHuang:
       I can't understand this code as follows:

    def tracking_callback(self, msg):
        self.track_flag = msg.flag1
        self.cx = msg.x
        self.cy = msg.y
        self.error_x = msg.error_x
        self.error_y = msg.error_y
        if len(self.pointx)>9:
            self.track_flag = True
        if self.phase == 2:
            self.track_flag = False
            self.phase = 1
        if (self.track_flag and -0.6 < self.waypoints[0].position.x and self.waypoints[0].position.x < 0.6):
            self.execute()
            self.default_pose_flag = False
        else:
            if not self.default_pose_flag:
                self.track_flag = False
                self.execute()
                self.default_pose_flag = True

    def execute(self):
        if self.track_flag:
            # Get the current pose so we can add it as a waypoint
            start_pose = self.arm.get_current_pose(self.end_effector_link).pose
            # Initialize the waypoints list
            self.waypoints= []
            # Set the first waypoint to be the starting pose
            # Append the pose to the waypoints list
            wpose = deepcopy(start_pose)
            # wpose.position.x = -0.5215
            # wpose.position.y = 0.2014
            # wpose.position.z = 0.4102
            if len(self.pointx)>8:
                if len(self.pointx)==9:
                    x_speed = np.mean(np.asarray(self.pointx[4:8]) - np.asarray(self.pointx[3:7]))
                    wpose.position.x += 2 * x_speed
                    wpose.position.z = 0.05
  
       what's mean of the self.phase and the self.pointsx. Ａnd in the tracking_callback(), Tｈere are four "if" statements,I can't understand ｈow do these three conditional statements work?

Haoran-Zhao, 2018-09-29:
So if I remembered correct, it has two mode, phase 1 is to following the
detected object. While end effector goes down, the camera will lost vision
of the object. so the phase 2 is to using predicted velocity to keep
tracking object, which is calculated by take the mean of 9 positions. The
thrid if statement is to check if the object is in the reachable range, if
not it goes back to initial position. That’s the whole idea.

When Li wakes up, he probably give you more detail and the correct
explainations.

weiwuhuhu, 2018-09-29:
And what's mean of this " x_speed = np.mean(np.asarray(self.pointx[4:8]) - np.asarray(self.pointx[3:7]))"
in the execute function.

Haoran-Zhao, 2018-09-29:
Because we found that the object almost moves a long x-axis, so this part
is what I mention about take the mean and predict the constant velocity
along x-axis.

weiwuhuhu, 2018-09-29:
> Because we found that the object almost moves a long x-axis, so this part is what I mention about take the mean and predict the constant velocity along x-axis.
> -- Haoran Zhao Graduate Research Assistant Electrical & Computer Engineering, University of Houston

Thank you very much! Your reply ｉｓ very great help to me.