lihuang3/ur5_ROS-Gazebo #9: catkin_make failed

williamjames88, 2018-12-31:
Hi, I'm getting this error whenever I try to build the file:

[ 57%] Building CXX object ur5_ROS-Gazebo/CMakeFiles/blocks_poses_publisher.dir/blocks_poses_publisher.cpp.o
/home/gg/ur_ws/src/ur5_ROS-Gazebo/blocks_poses_publisher.cpp:14:39: fatal error: ur5_notebook/blocks_poses.h: No such file or directory
compilation terminated.
ur5_ROS-Gazebo/CMakeFiles/blocks_poses_publisher.dir/build.make:62: recipe for target 'ur5_ROS-Gazebo/CMakeFiles/blocks_poses_publisher.dir/blocks_poses_publisher.cpp.o' failed
make[2]: *** [ur5_ROS-Gazebo/CMakeFiles/blocks_poses_publisher.dir/blocks_poses_publisher.cpp.o] Error 1
CMakeFiles/Makefile2:1747: recipe for target 'ur5_ROS-Gazebo/CMakeFiles/blocks_poses_publisher.dir/all' failed
make[1]: *** [ur5_ROS-Gazebo/CMakeFiles/blocks_poses_publisher.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j1 -l1" failed

This occurred after I have completed this part and perform catkin_make:

In the same directory, make a copy of common.gazebo.xacro and ur5.urdf.xacro in case of any malfunction. These two default files do not include camera and vacuum gripper modules. So we would replace these two files with customized files. Under directory ur_ws/src/ur5_ROS-Gazebo/src/ur_description/, copy common.gazebo.xacro and ur5.urdf.xacro to ur_ws/src/universal_robot/ur_description/urdf/.

Can someone please help me out?
I need to get the installation done asap.

lihuang3, 2019-01-01:
@williamjames88 I would suggest you undo this step and see if you could catkin make without errors. Meanwhile, please double check whether you had followed the instruction step-by-step.

williamjames88, 2019-01-01:
@lihuang3 Hi, I heeded your advice and I'm still getting the same error. 
I also tried creating a new folder and downloaded only your repository into it but still fails to catkin_make.
Here's the full code: 

gg@gg:~/ws$ catkin_make
Base path: /home/gg/ws
Source space: /home/gg/ws/src
Build space: /home/gg/ws/build
Devel space: /home/gg/ws/devel
Install space: /home/gg/ws/install
Creating symlink "/home/gg/ws/src/CMakeLists.txt" pointing to "/opt/ros/kinetic/share/catkin/cmake/toplevel.cmake"
####
#### Running command: "cmake /home/gg/ws/src -DCATKIN_DEVEL_PREFIX=/home/gg/ws/devel -DCMAKE_INSTALL_PREFIX=/home/gg/ws/install -G Unix Makefiles" in "/home/gg/ws/build"
####
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/gg/ws/devel
-- Using CMAKE_PREFIX_PATH: /home/gg/ur5_ws/devel;/home/gg/ws_moveit/devel;/opt/ros/kinetic
-- This workspace overlays: /home/gg/ws_moveit/devel;/opt/ros/kinetic
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/gg/ws/build/test_results
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.14
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - ur5_notebook
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'ur5_notebook'
-- ==> add_subdirectory(ur5_ROS-Gazebo)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- ur5_notebook: 2 messages, 0 services
-- Configuring done
-- Generating done
-- Build files have been written to: /home/gg/ws/build
####
#### Running command: "make -j1 -l1" in "/home/gg/ws/build"
####
Scanning dependencies of target blocks_poses_publisher
[  6%] Building CXX object ur5_ROS-Gazebo/CMakeFiles/blocks_poses_publisher.dir/blocks_poses_publisher.cpp.o
/home/gg/ws/src/ur5_ROS-Gazebo/blocks_poses_publisher.cpp:14:39: fatal error: ur5_notebook/blocks_poses.h: No such file or directory
compilation terminated.
ur5_ROS-Gazebo/CMakeFiles/blocks_poses_publisher.dir/build.make:62: recipe for target 'ur5_ROS-Gazebo/CMakeFiles/blocks_poses_publisher.dir/blocks_poses_publisher.cpp.o' failed
make[2]: *** [ur5_ROS-Gazebo/CMakeFiles/blocks_poses_publisher.dir/blocks_poses_publisher.cpp.o] Error 1
CMakeFiles/Makefile2:566: recipe for target 'ur5_ROS-Gazebo/CMakeFiles/blocks_poses_publisher.dir/all' failed
make[1]: *** [ur5_ROS-Gazebo/CMakeFiles/blocks_poses_publisher.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j1 -l1" failed

lihuang3, 2019-01-22:
@williamjames88  Looks like that there is a `blocks_poses.h` file under 'ur5_ws/devel/include/ur5_notebook/', which I assume it was not configured by ourselves. Anyway, I just push the file to the root directory of the repo. You can try adding it to the directory mentioned above.

RDaneelOlivav, 2019-03-25:
It worked the compilation for me. The procedure I followed was:
1) Start from a clean ~/catkin_ws
2) cd ~/catkin_ws/src; git clone https://github.com/ros-industrial/universal_robot.git
3) cd ~/catkin_ws; catkin_make; source devel/setup.bash
4) cd ~/catkin_ws/src; git clone https://github.com/lihuang3/ur5_ROS-Gazebo.git
5) cp ~/catkin_ws/src/ur5_ROS-Gazebo/src/ur_description/ur5.urdf.xacro ~/catkin_ws/src/universal_robot/ur_description/urdf/
6) cp ~/catkin_ws/src/ur5_ROS-Gazebo/src/ur_description/common.gazebo.xacro ~/catkin_ws/src/universal_robot/ur_description/urdf/
7) cp ~/catkin_ws/src/ur5_ROS-Gazebo/blocks_poses.h ~/catkin_ws/devel/include/ur5_notebook/

Hope it helps, I'll continue testing ;)

portgasray, 2019-06-08:
It works for me:

[image]

yyc9268, 2020-01-09:
I found a generally working solution.
Make the include directory structure as 'ur5_ROS-Gazebo/include/ur5_notebook'
And move the 'blocks_poses.h' into this.
The final structure should be 'ur5_ROS-Gazebo/include/ur5_notebook/blocks_poses.h'
This solution clearly works for both of 'catkin make' and 'colcon build'.

Cyano0, 2020-05-18:
I think I have this similar issue after I tried everything mentioned above, done adding blocks_poses.h to 'include/ur5_notebook' directory. Unfortunately there is still error when dong catkin_make. Could anyone be able to help by any chance? 

Here is the error information:

[  0%] Built target _ur_msgs_generate_messages_check_deps_SetIO
[  0%] Built target _ur_msgs_generate_messages_check_deps_Analog
[  0%] Built target _ur_msgs_generate_messages_check_deps_ToolDataMsg
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target _ur_msgs_generate_messages_check_deps_IOStates
[  0%] Built target _ur_msgs_generate_messages_check_deps_RobotStateRTMsg
[  0%] Built target _ur_msgs_generate_messages_check_deps_MasterboardDataMsg
[  0%] Built target _ur_msgs_generate_messages_check_deps_Digital
[  0%] Built target _ur_msgs_generate_messages_check_deps_SetSpeedSliderFraction
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target _ur_msgs_generate_messages_check_deps_SetPayload
[  0%] Built target _ur_msgs_generate_messages_check_deps_RobotModeDataMsg
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target gazebo_msgs_generate_messages_lisp
[  0%] Built target sensor_msgs_generate_messages_py
[  0%] Built target std_srvs_generate_messages_eus
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target gazebo_msgs_generate_messages_nodejs
[  0%] Built target gazebo_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_eus
[  0%] Built target roscpp_generate_messages_cpp
[  0%] Built target sensor_msgs_generate_messages_nodejs
[  0%] Built target trajectory_msgs_generate_messages_lisp
[  0%] Built target gazebo_msgs_generate_messages_py
[  0%] Built target trajectory_msgs_generate_messages_py
[  0%] Built target sensor_msgs_generate_messages_eus
[  0%] Built target trajectory_msgs_generate_messages_eus
[  0%] Built target sensor_msgs_generate_messages_cpp
[  0%] Built target trajectory_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target std_srvs_generate_messages_cpp
[  0%] Built target roscpp_generate_messages_nodejs
[  0%] Built target rosgraph_msgs_generate_messages_lisp
[  0%] Built target rosgraph_msgs_generate_messages_py
[  0%] Built target trajectory_msgs_generate_messages_nodejs
[  0%] Built target std_srvs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target roscpp_generate_messages_lisp
[  0%] Built target std_srvs_generate_messages_py
[  0%] Built target roscpp_generate_messages_eus
[  0%] Built target roscpp_generate_messages_py
[  0%] Built target rosgraph_msgs_generate_messages_cpp
[  0%] Built target gazebo_msgs_generate_messages_eus
[  0%] Built target rosgraph_msgs_generate_messages_eus
[  0%] Built target std_srvs_generate_messages_nodejs
[  0%] Built target rosgraph_msgs_generate_messages_nodejs
[  0%] Built target geometry_msgs_generate_messages_nodejs
[  0%] Built target sensor_msgs_generate_messages_lisp
[  1%] Linking CXX executable /home/interact/ur_ws/devel/lib/ur5_notebook/blocks_spawner
[  2%] Built target ur_driver_gencfg
[  2%] Built target _ur5_notebook_generate_messages_check_deps_Tracker
[  4%] Built target ur10_kin
[  4%] Built target _ur5_notebook_generate_messages_check_deps_blocks_poses
[  9%] Built target ur3_kin
[  9%] Built target ur5_kin
[ 21%] Built target ur_msgs_generate_messages_nodejs
[ 34%] Built target ur_msgs_generate_messages_cpp
[ 48%] Built target ur_msgs_generate_messages_py
[ 60%] Built target ur_msgs_generate_messages_lisp
[ 74%] Built target ur_msgs_generate_messages_eus
[ 75%] Linking CXX executable /home/interact/ur_ws/devel/lib/ur5_notebook/blocks_poses_publisher
[ 79%] Built target ur5_notebook_generate_messages_py
[ 81%] Built target ur5_notebook_generate_messages_nodejs
[ 84%] Built target ur5_notebook_generate_messages_cpp
[ 87%] Built target ur5_notebook_generate_messages_eus
ur5_ROS-Gazebo/CMakeFiles/blocks_spawner.dir/build.make:113: recipe for target '/home/interact/ur_ws/devel/lib/ur5_notebook/blocks_spawner' failed
make[2]: *** [/home/interact/ur_ws/devel/lib/ur5_notebook/blocks_spawner] Error 1
CMakeFiles/Makefile2:2035: recipe for target 'ur5_ROS-Gazebo/CMakeFiles/blocks_spawner.dir/all' failed
make[1]: *** [ur5_ROS-Gazebo/CMakeFiles/blocks_spawner.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 92%] Built target ur10_moveit_plugin
[ 95%] Built target ur5_moveit_plugin
collect2: error: ld returned 1 exit status
ur5_ROS-Gazebo/CMakeFiles/blocks_poses_publisher.dir/build.make:113: recipe for target '/home/interact/ur_ws/devel/lib/ur5_notebook/blocks_poses_publisher' failed
make[2]: *** [/home/interact/ur_ws/devel/lib/ur5_notebook/blocks_poses_publisher] Error 1
CMakeFiles/Makefile2:1902: recipe for target 'ur5_ROS-Gazebo/CMakeFiles/blocks_poses_publisher.dir/all' failed
make[1]: *** [ur5_ROS-Gazebo/CMakeFiles/blocks_poses_publisher.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2