lihuang3/ur5_ROS-Gazebo #10: which file should I execute to control the robot arm to grasp the red box

aositeluofu, 2019-03-15:
Hi,lihuang3.Thank you very much for your excellent work.Still,I would like to know how to let the robot arm move to grasp the red box after the "roslaunch ur5_notebook initialize.launch".
I am still a rockie to this ROS area.Can you tell me which file should I execute? Looking forward to your reply.Thank you very much

lihuang3, 2019-03-15:
Hi @aositeluofu, if you strictly follow the instructions, the robot arm should do the pick and place once you `roslaunch ur5_notebook initialize.launch`. You may find this discussion helpful https://github.com/lihuang3/ur5_ROS-Gazebo/issues/3#issue-359420939

aositeluofu, 2019-03-16:
thank you very much.it's very helpful of you.may you good luck in your life

hxn8439, 2021-09-04:
I have the same problem too. The robot arm suppose to move after executing
roslaunch ur5_notebook initialize.launch

So, I also execute python testvisionplanner.py and testmove2.py but endup with this error return.

[ERROR] [1630771695.765029]: Unable to immediately register with master node [http://localhost:11311]: master may not be running yet. Will keep trying.

so the error is happening at the master node communication.