lihuang3/ur5_ROS-Gazebo #12: [rospack] Error: package 'ur5_ROS-Gazebo' not found

IndustrialEngStudent, 2019-05-06:
Hi!

I am having problems running testmotion.py.
This is what happens:
```

laurah@laurah-HP-EliteBook-840-G1:~/catkin_ws$ catkin_make
Base path: /home/laurah/catkin_ws
Source space: /home/laurah/catkin_ws/src
Build space: /home/laurah/catkin_ws/build
Devel space: /home/laurah/catkin_ws/devel
Install space: /home/laurah/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/laurah/catkin_ws/build"
####
####
#### Running command: "make -j4 -l4" in "/home/laurah/catkin_ws/build"
####
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target _ur_msgs_generate_messages_check_deps_IOStates
[  0%] Built target _ur_msgs_generate_messages_check_deps_MasterboardDataMsg
[  0%] Built target _ur_msgs_generate_messages_check_deps_Analog
[  0%] Built target _ur_msgs_generate_messages_check_deps_RobotModeDataMsg
[  0%] Built target _ur_msgs_generate_messages_check_deps_Digital
[  0%] Built target _ur_msgs_generate_messages_check_deps_ToolDataMsg
[  0%] Built target _ur_msgs_generate_messages_check_deps_SetIO
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target _ur_msgs_generate_messages_check_deps_RobotStateRTMsg
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target _ur_msgs_generate_messages_check_deps_SetPayload
[  1%] Built target blocks_poses_publisher
[  3%] Built target blocks_spawner
[  5%] Built target ur_driver_gencfg
[  5%] Built target industrial_msgs_generate_messages_lisp
[  5%] Built target _ur5_notebook_generate_messages_check_deps_Tracker
[  5%] Built target _ur5_notebook_generate_messages_check_deps_blocks_poses
[  5%] Built target _catkin_empty_exported_target
[  5%] Built target sensor_msgs_generate_messages_py
[  5%] Built target std_srvs_generate_messages_eus
[  5%] Built target geometry_msgs_generate_messages_py
[  5%] Built target geometry_msgs_generate_messages_eus
[  5%] Built target roscpp_generate_messages_cpp
[  5%] Built target sensor_msgs_generate_messages_nodejs
[  5%] Built target sensor_msgs_generate_messages_cpp
[  5%] Built target trajectory_msgs_generate_messages_lisp
[  5%] Built target trajectory_msgs_generate_messages_eus
[  5%] Built target trajectory_msgs_generate_messages_py
[  5%] Built target sensor_msgs_generate_messages_eus
[  5%] Built target geometry_msgs_generate_messages_cpp
[  5%] Built target std_srvs_generate_messages_cpp
[  5%] Built target trajectory_msgs_generate_messages_cpp
[  5%] Built target rosgraph_msgs_generate_messages_lisp
[  5%] Built target roscpp_generate_messages_nodejs
[  5%] Built target rosgraph_msgs_generate_messages_py
[  5%] Built target trajectory_msgs_generate_messages_nodejs
[  5%] Built target geometry_msgs_generate_messages_lisp
[  5%] Built target std_srvs_generate_messages_lisp
[  5%] Built target std_srvs_generate_messages_py
[  5%] Built target roscpp_generate_messages_lisp
[  5%] Built target roscpp_generate_messages_eus
[  5%] Built target roscpp_generate_messages_py
[  5%] Built target rosgraph_msgs_generate_messages_cpp
[  5%] Built target rosgraph_msgs_generate_messages_eus
[  5%] Built target std_srvs_generate_messages_nodejs
[  5%] Built target rosgraph_msgs_generate_messages_nodejs
[  5%] Built target geometry_msgs_generate_messages_nodejs
[  5%] Built target sensor_msgs_generate_messages_lisp
[  5%] Built target industrial_msgs_generate_messages_cpp
[  5%] Built target controller_manager_msgs_generate_messages_nodejs
[  5%] Built target control_msgs_generate_messages_nodejs
[  5%] Built target tf_generate_messages_eus
[  5%] Built target control_msgs_generate_messages_py
[  5%] Built target actionlib_msgs_generate_messages_py
[  5%] Built target actionlib_msgs_generate_messages_lisp
[  5%] Built target control_msgs_generate_messages_eus
[  5%] Built target actionlib_msgs_generate_messages_eus
[  5%] Built target industrial_msgs_generate_messages_eus
[  5%] Built target actionlib_generate_messages_lisp
[  5%] Built target actionlib_generate_messages_eus
[  5%] Built target tf_generate_messages_cpp
[  5%] Built target controller_manager_msgs_generate_messages_cpp
[  5%] Built target controller_manager_msgs_generate_messages_eus
[  5%] Built target tf2_msgs_generate_messages_eus
[  5%] Built target actionlib_generate_messages_py
[  5%] Built target actionlib_generate_messages_cpp
[  5%] Built target tf2_msgs_generate_messages_py
[  5%] Built target control_msgs_generate_messages_lisp
[  5%] Built target actionlib_msgs_generate_messages_nodejs
[  5%] Built target controller_manager_msgs_generate_messages_lisp
[  5%] Built target control_msgs_generate_messages_cpp
[  5%] Built target industrial_msgs_generate_messages_py
[  5%] Built target industrial_msgs_generate_messages_nodejs
[  5%] Built target tf_generate_messages_lisp
[  5%] Built target tf2_msgs_generate_messages_nodejs
[  5%] Built target tf_generate_messages_nodejs
[  5%] Built target tf_generate_messages_py
[  5%] Built target controller_manager_msgs_generate_messages_py
[  5%] Built target tf2_msgs_generate_messages_lisp
[  5%] Built target tf2_msgs_generate_messages_cpp
[  7%] Built target ur_hardware_interface
[  7%] Built target actionlib_generate_messages_nodejs
[  7%] Built target actionlib_msgs_generate_messages_cpp
[ 10%] Built target ur10_kin
[ 13%] Built target ur5_kin
[ 15%] Built target ur3_kin
[ 25%] Built target ur_msgs_generate_messages_cpp
[ 34%] Built target ur_msgs_generate_messages_nodejs
[ 46%] Built target ur_msgs_generate_messages_py
[ 55%] Built target ur_msgs_generate_messages_lisp
[ 65%] Built target ur_msgs_generate_messages_eus
[ 68%] Built target ur5_notebook_generate_messages_nodejs
[ 71%] Built target ur5_notebook_generate_messages_py
[ 72%] Built target ur5_notebook_generate_messages_cpp
[ 73%] Built target ur5_notebook_generate_messages_lisp
[ 77%] Built target ur5_notebook_generate_messages_eus
[ 78%] Built target ur10_moveit_plugin
[ 96%] Built target ur_driver
[ 98%] Built target ur5_moveit_plugin
[100%] Built target ur3_moveit_plugin
[100%] Built target ur_msgs_generate_messages
[100%] Built target ur5_notebook_generate_messages
laurah@laurah-HP-EliteBook-840-G1:~/catkin_ws$ source devel/setup.bash
laurah@laurah-HP-EliteBook-840-G1:~/catkin_ws$ rosrun ur5_ROS-Gazebo testmotion.py
[rospack] Error: package 'ur5_ROS-Gazebo' not found
laurah@laurah-HP-EliteBook-840-G1:~/catkin_ws$ 

```
I installed the package in my catkin_ws/src folder using "git clone https://github.com/lihuang3/ur5_ROS-Gazebo.git" 

Why won't it find the package?

lihuang3, 2019-05-08:
@IndustrialEngStudent Did you try `roslaunch ur5_notebook initialize.launch`

IndustrialEngStudent, 2019-05-09:
Yes, it works! I did not notice that the actual package name was ur5_notebook. Thank you

zkgoodboy, 2019-05-09:
> Yes, it works! I did not notice that the actual package name was ur5_notebook. Thank you

do you make it work？ may i ask you for some help?,i don't make it wok until now.so frustrated now!
looking forward to your help!

closed 2019-06-04 by IndustrialEngStudent