lihuang3/ur5_ROS-Gazebo #13: error  when roslunch ur5_notebook initialize.launch

zkgoodboy, 2019-05-08:
hello,lihuang,when i replace these two files with customized files. Under directory ur_ws/src/ur5_ROS-Gazebo/src/ur_description/, copy common.gazebo.xacro and ur5.urdf.xacro to ur_ws/src/universal_robot/ur_description/urdf/.then i roslunch ur5_notebook initialize.launch,there ia an error makes can not start the gazebo,i am so confused,would you help me,folling is the detail about error:

###(ERROR) Invalid parameter "transmission_hw_interface" None ### None
when instantiating macro: ur5_robot (/home/hourui/catkin_ws/src/universal_robot-kinetic-devel/universal_robot-kinetic-devel/ur_description/urdf/ur5.urdf.xacro)
in file: /home/hourui/catkin_ws/src/universal_robot-kinetic-devel/universal_robot-kinetic-devel/ur_description/urdf/ur5_joint_limited_robot.urdf.xacro
### {ERROR}while processing /home/hourui/catkin_ws/src/universal_robot-kinetic-devel/universal_robot-kinetic-devel/ur_description/launch/ur5_upload.launch:
Invalid <param> tag: Cannot load command parameter [robot_description]: command [/opt/ros/kinetic/lib/xacro/xacro --inorder '/home/hourui/catkin_ws/src/universal_robot-kinetic-devel/universal_robot-kinetic-devel/ur_description/urdf/ur5_joint_limited_robot.urdf.xacro' transmission_hw_interface:=hardware_interface/PositionJointInterface] returned with code [2]. 

Param xml is <param command="$(find xacro)/xacro --inorder '$(find ur_description)/urdf/ur5_joint_limited_robot.urdf.xacro' transmission_hw_interface:=$(arg transmission_hw_interface)" if="$(arg limited)" name="robot_description"/>
The traceback for the exception was written to the log file
### ### 
waiting for your help,grateful about you!

lihuang3, 2019-05-08:
@zkgoodboy It seems that the error indicates some issues with the joint limit. Did you change the joint limit in `ur5_joint_limited_robot.urdf.xacro`?
```
  shoulder_pan_lower_limit="${-2*pi}" shoulder_pan_upper_limit="${2*pi}"
```

zkgoodboy, 2019-05-08:
> @zkgoodboy It seems that the error indicates some issues with the joint limit. Did you change the joint limit in `ur5_joint_limited_robot.urdf.xacro`?
> 
> ```
>   shoulder_pan_lower_limit="${-2*pi}" shoulder_pan_upper_limit="${2*pi}"
> ```

i did,if i didn,t repace these two files before ,when i roslunch ur5_notebook initialize.launch,i can turn on the gazebo,but no gripper and usb cam.after i replace them,here is the error,and the gazebo didn't work.
for this error,i work the whole night until now,so upset!

lihuang3, 2019-05-08:
@zkgoodboy Looks like you are using `kinetic-devel` version of Universal Robot, but this repo was based on `kinetic` version. This might be a possible reason.

zkgoodboy, 2019-05-08:
i try it again with the another version of universal robot,but the error still exist,just change the path's name.following is the error:
**Invalid parameter "transmission_hw_interface" None None**
when instantiating macro: ur5_robot (/home/hourui/catkin_ws/src/universal_robot/ur_description/urdf/ur5.urdf.xacro)
in file: /home/hourui/catkin_ws/src/universal_robot/ur_description/urdf/ur5_joint_limited_robot.urdf.xacro
### while processing /home/hourui/catkin_ws/src/universal_robot/ur_description/launch/ur5_upload.launch:
Invalid <param> tag: Cannot load command parameter [robot_description]: command [/opt/ros/kinetic/lib/xacro/xacro --inorder '/home/hourui/catkin_ws/src/universal_robot/ur_description/urdf/ur5_joint_limited_robot.urdf.xacro' transmission_hw_interface:=hardware_interface/PositionJointInterface] returned with code [2]. 

### Param xml is <param command="$(find xacro)/xacro --inorder '$(find ur_description)/urdf/ur5_joint_limited_robot.urdf.xacro' transmission_hw_interface:=$(arg transmission_hw_interface)" if="$(arg limited)" name="robot_description"/>
The traceback for the exception was written to the log file

can i add your wechat?i really want fix it with your help!

zkgoodboy, 2019-05-08:
my wechat :18428341003
grateful for your help!

zkgoodboy, 2019-05-09:
hello li huang, how do i Make sure i have installed Python2.7 and some useful libraries/packages, such as Numpy, cv2, etc.(i  guess mabey it is the problem,or not)

lihuang3, 2019-05-09:
@zkgoodboy check this most recent issue `https://github.com/lihuang3/ur5_ROS-Gazebo/issues/12#issuecomment-490756615`, you may ask him/her for details as he/she just made it work. Meanwhile, the recommended environment should have Ubuntu 16.04, Python 2.7, ROS kinetic installed. For any other missing packages, there should be some explicit error message that reminds you of installation.

zkgoodboy, 2019-05-09:
> @zkgoodboy check this most recent [issue](https://github.com/lihuang3/ur5_ROS-Gazebo/issues/12#issuecomment-490756615), you may ask him/her for details as he/she just made it work. Meanwhile, the [...]

thanks for your help.i am really appreciated about it .i got a new issue:my moveit setup assistant can't load my urdf robot.(is  it resulted the error?)

lihuang3, 2019-05-09:
@zkgoodboy That might be an issue. Before you roslaunch the program, test UR5 in Moveit. If there are errors with Moveit, try rebooting your laptop or reinstalling Moveit.

zkgoodboy, 2019-05-11:
hello, lihuang3,i  have a question.after i replace the ur5.urdf.xacro under the universal robot folder with the ur5.urdf.xacro under your repo. Do i need to roslaunch moveit setup assistant to regenerate a new moveit_config files and replace the ur5_moveit_config files before(under the universal robot) ？because we add the gripper and usb cam in our new urdf.

lihuang3, 2019-05-11:
@zkgoodboy Not necessary. Just do `catkin make`.

strowlands, 2019-05-29:
Hello, I followed the instructions in the readme exactly and am getting the same error. Was there any resolution to this?

ghx-0228, 2019-06-11:
Hi, Guys
I have also encountered two above problems. Could you share your solutions?

roboconlab, 2019-06-19:
hello,i met follow question,could you help me?

Invalid parameter "transmission_hw_interface" None None
when instantiating macro: ur5_robot (/home/c411/ur_ws/src/universal_robot/ur_description/urdf/ur5.urdf.xacro)
in file: /home/c411/ur_ws/src/universal_robot/ur_description/urdf/ur5_joint_limited_robot.urdf.xacro
while processing /home/c411/ur_ws/src/universal_robot/ur_description/launch/ur5_upload.launch:
Invalid <param> tag: Cannot load command parameter [robot_description]: command [/opt/ros/kinetic/lib/xacro/xacro --inorder '/home/c411/ur_ws/src/universal_robot/ur_description/urdf/ur5_joint_limited_robot.urdf.xacro'] returned with code [2]. 

Param xml is <param command="$(find xacro)/xacro --inorder '$(find ur_description)/urdf/ur5_joint_limited_robot.urdf.xacro'" if="$(arg limited)" name="robot_description"/>
The traceback for the exception was written to the log file

roboconlab, 2019-06-27:
question<1>～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～
Invalid parameter "transmission_hw_interface" None None
when instantiating macro: ur5_robot (/home/c411/ur_ws/src/universal_robot/ur_description/urdf/ur5.urdf.xacro)
in file: /home/c411/ur_ws/src/universal_robot/ur_description/urdf/ur5_joint_limited_robot.urdf.xacro
while processing /home/c411/ur_ws/src/universal_robot/ur_description/launch/ur5_upload.launch:
Invalid tag: Cannot load command parameter [robot_description]: command [/opt/ros/kinetic/lib/xacro/xacro --inorder '/home/c411/ur_ws/src/universal_robot/ur_description/urdf/ur5_joint_limited_robot.urdf.xacro'] returned with code [2].
解决方法：将transmission_hw_interface添加到ur5.urdf.xacro中的xacro函数的参数申明中，即：
  <xacro:macro name="ur5_robot" params="prefix joint_limited
     shoulder_pan_lower_limit:=${-pi}    shoulder_pan_upper_limit:=${pi}
     shoulder_lift_lower_limit:=${-pi}    shoulder_lift_upper_limit:=${pi}
     elbow_joint_lower_limit:=${-pi}    elbow_joint_upper_limit:=${pi}
     wrist_1_lower_limit:=${-pi}    wrist_1_upper_limit:=${pi}
     wrist_2_lower_limit:=${-pi}    wrist_2_upper_limit:=${pi}
     wrist_3_lower_limit:=${-pi}    wrist_3_upper_limit:=${pi}
     transmission_hw_interface">

问题<2>～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～
Undefined parameters [hw_interface] None None
when instantiating macro: ur_arm_transmission (/home/c411/ur_ws/src/universal_robot/ur_description/urdf/ur.transmission.xacro)
instantiated from: ur5_robot (/home/c411/ur_ws/src/universal_robot/ur_description/urdf/ur5.urdf.xacro)
in file: /home/c411/ur_ws/src/universal_robot/ur_description/urdf/ur5_joint_limited_robot.urdf.xacro
while processing /home/c411/ur_ws/src/universal_robot/ur_description/launch/ur5_upload.launch:
Invalid <param> tag: Cannot load command parameter [robot_description]: command [/opt/ros/kinetic/lib/xacro/xacro --inorder '/home/c411/ur_ws/src/universal_robot/ur_description/urdf/ur5_joint_limited_robot.urdf.xacro' transmission_hw_interface:=hardware_interface/PositionJointInterface] returned with code [2]. 
解决方法：将hw_interface="${transmission_hw_interface}"添加到在ur5.urdf.xacro中实例化时的<xacro:ur_arm_transmission prefix="${prefix}"/>中，即：<xacro:ur_arm_transmission prefix="${prefix}" hw_interface="${transmission_hw_interface}"/>

问题<3>～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～
[ERROR] [1561634513.358761694, 0.174000000]: GazeboRosControlPlugin missing <legacyModeNS> while using DefaultRobotHWSim, defaults to true.
This setting assumes you have an old package with an old implementation of DefaultRobotHWSim, where the robotNamespace is disregarded and absolute paths are used instead.
If you do not want to fix this issue in an old package just set <legacyModeNS> to true.
解决方法：将<legacyModeNS>true</legacyModeNS>添加到common.gazebo.xacro文件中，即：  
<gazebo>
    <plugin name="ros_control" filename="libgazebo_ros_control.so">
    <legacyModeNS>true</legacyModeNS>
      
      
    </plugin>   	
</gazebo>

roboconlab, 2019-06-27:
问题<1>～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～
Invalid parameter "transmission_hw_interface" None None
when instantiating macro: ur5_robot (/home/c411/ur_ws/src/universal_robot/ur_description/urdf/ur5.urdf.xacro)
in file: /home/c411/ur_ws/src/universal_robot/ur_description/urdf/ur5_joint_limited_robot.urdf.xacro
while processing /home/c411/ur_ws/src/universal_robot/ur_description/launch/ur5_upload.launch:
Invalid tag: Cannot load command parameter [robot_description]: command [/opt/ros/kinetic/lib/xacro/xacro --inorder '/home/c411/ur_ws/src/universal_robot/ur_description/urdf/ur5_joint_limited_robot.urdf.xacro'] returned with code [2].
解决方法：将transmission_hw_interface添加到ur5.urdf.xacro中的xacro函数的参数申明中，即：
  <xacro:macro name="ur5_robot" params="prefix joint_limited
     shoulder_pan_lower_limit:=${-pi}    shoulder_pan_upper_limit:=${pi}
     shoulder_lift_lower_limit:=${-pi}    shoulder_lift_upper_limit:=${pi}
     elbow_joint_lower_limit:=${-pi}    elbow_joint_upper_limit:=${pi}
     wrist_1_lower_limit:=${-pi}    wrist_1_upper_limit:=${pi}
     wrist_2_lower_limit:=${-pi}    wrist_2_upper_limit:=${pi}
     wrist_3_lower_limit:=${-pi}    wrist_3_upper_limit:=${pi}
     transmission_hw_interface">

问题<2>～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～
Undefined parameters [hw_interface] None None
when instantiating macro: ur_arm_transmission (/home/c411/ur_ws/src/universal_robot/ur_description/urdf/ur.transmission.xacro)
instantiated from: ur5_robot (/home/c411/ur_ws/src/universal_robot/ur_description/urdf/ur5.urdf.xacro)
in file: /home/c411/ur_ws/src/universal_robot/ur_description/urdf/ur5_joint_limited_robot.urdf.xacro
while processing /home/c411/ur_ws/src/universal_robot/ur_description/launch/ur5_upload.launch:
Invalid <param> tag: Cannot load command parameter [robot_description]: command [/opt/ros/kinetic/lib/xacro/xacro --inorder '/home/c411/ur_ws/src/universal_robot/ur_description/urdf/ur5_joint_limited_robot.urdf.xacro' transmission_hw_interface:=hardware_interface/PositionJointInterface] returned with code [2]. 
解决方法：将hw_interface="${transmission_hw_interface}"添加到在ur5.urdf.xacro中实例化时的<xacro:ur_arm_transmission prefix="${prefix}"/>中，即：<xacro:ur_arm_transmission prefix="${prefix}" hw_interface="${transmission_hw_interface}"/>

问题<3>～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～
[ERROR] [1561634513.358761694, 0.174000000]: GazeboRosControlPlugin missing <legacyModeNS> while using DefaultRobotHWSim, defaults to true.
This setting assumes you have an old package with an old implementation of DefaultRobotHWSim, where the robotNamespace is disregarded and absolute paths are used instead.
If you do not want to fix this issue in an old package just set <legacyModeNS> to true.
解决方法：将<legacyModeNS>true</legacyModeNS>添加到common.gazebo.xacro文件中，即：  
<gazebo>
    <plugin name="ros_control" filename="libgazebo_ros_control.so">
    <legacyModeNS>true</legacyModeNS>
      
      
    </plugin>   	
</gazebo>

roboconlab, 2019-06-27:
问题<3>～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～
[ERROR] [1561634513.358761694, 0.174000000]: GazeboRosControlPlugin missing <legacyModeNS> while using DefaultRobotHWSim, defaults to true.
This setting assumes you have an old package with an old implementation of DefaultRobotHWSim, where the robotNamespace is disregarded and absolute paths are used instead.
If you do not want to fix this issue in an old package just set <legacyModeNS> to true.
解决方法：将<legacyModeNS>true</legacyModeNS>添加到common.gazebo.xacro文件中，即：  
<gazebo>
    <plugin name="ros_control" filename="libgazebo_ros_control.so">
    <legacyModeNS>true</legacyModeNS>
      
      
    </plugin>   	
</gazebo>

Diankuang-Wu, 2019-07-02:
> 问题<1>～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～
> Invalid parameter "transmission_hw_interface" None None
> when instantiating macro: ur5_robot (/home/c411/ur_ws/src/universal_robot/ur_description/urdf/ur5.urdf.xacro)
> in file: /home/c411/ur_ws/src/universal_robot/ur_description/urdf/ur5_joint_limited_robot.urdf.xacro
> while processing /home/c411/ur_ws/src/universal_robot/ur_description/launch/ur5_upload.launch:
> Invalid tag: Cannot load command parameter [robot_description]: command [/opt/ros/kinetic/lib/xacro/xacro --inorder '/home/c411/ur_ws/src/universal_robot/ur_description/urdf/ur5_joint_limited_robot.urdf.xacro'] returned with code [2].
> 解决方法：将transmission_hw_interface添加到ur5.urdf.xacro中的xacro函数的参数申明中，即：
> <xacro:macro name="ur5_robot" params="prefix joint_limited
> shoulder_pan_lower_limit:=${-pi} shoulder_pan_upper_limit:=${pi}
> shoulder_lift_lower_limit:=${-pi} shoulder_lift_upper_limit:=${pi}
> elbow_joint_lower_limit:=${-pi} elbow_joint_upper_limit:=${pi}
> wrist_1_lower_limit:=${-pi} wrist_1_upper_limit:=${pi}
> wrist_2_lower_limit:=${-pi} wrist_2_upper_limit:=${pi}
> wrist_3_lower_limit:=${-pi} wrist_3_upper_limit:=${pi}
> transmission_hw_interface">
> 
> 问题<2>～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～
> Undefined parameters [hw_interface] None None
> when instantiating macro: ur_arm_transmission (/home/c411/ur_ws/src/universal_robot/ur_description/urdf/ur.transmission.xacro)
> instantiated from: ur5_robot (/home/c411/ur_ws/src/universal_robot/ur_description/urdf/ur5.urdf.xacro)
> in file: /home/c411/ur_ws/src/universal_robot/ur_description/urdf/ur5_joint_limited_robot.urdf.xacro
> while processing /home/c411/ur_ws/src/universal_robot/ur_description/launch/ur5_upload.launch:
> Invalid tag: Cannot load command parameter [robot_description]: command [/opt/ros/kinetic/lib/xacro/xacro --inorder '/home/c411/ur_ws/src/universal_robot/ur_description/urdf/ur5_joint_limited_robot.urdf.xacro' transmission_hw_interface:=hardware_interface/PositionJointInterface] returned with code [2].
> 解决方法：将hw_interface="${transmission_hw_interface}"添加到在ur5.urdf.xacro中实例化时的<xacro:ur_arm_transmission prefix="${prefix}"/>中，即：<xacro:ur_arm_transmission prefix="${prefix}" hw_interface="${transmission_hw_interface}"/>

nice! you are right, thx for your help.

clintonelian, 2019-08-29:
I have edited several lines in  ur5.urdf.xacro to make it works

before :
  <xacro:macro name="ur5_robot" params="prefix joint_limited
     shoulder_pan_lower_limit:=${-pi}    shoulder_pan_upper_limit:=${pi}
     shoulder_lift_lower_limit:=${-pi}    shoulder_lift_upper_limit:=${pi}
     elbow_joint_lower_limit:=${-pi}    elbow_joint_upper_limit:=${pi}
     wrist_1_lower_limit:=${-pi}    wrist_1_upper_limit:=${pi}
     wrist_2_lower_limit:=${-pi}    wrist_2_upper_limit:=${pi}
     wrist_3_lower_limit:=${-pi}    wrist_3_upper_limit:=${pi}">
after : 
  <xacro:macro name="ur5_robot" params="prefix joint_limited
     **shoulder_pan_lower_limit:=${-2*pi}    shoulder_pan_upper_limit:=${2*pi}**
     shoulder_lift_lower_limit:=${-pi}    shoulder_lift_upper_limit:=${pi}
     elbow_joint_lower_limit:=${-pi}    elbow_joint_upper_limit:=${pi}
     wrist_1_lower_limit:=${-pi}    wrist_1_upper_limit:=${pi}
     wrist_2_lower_limit:=${-pi}    wrist_2_upper_limit:=${pi}
     wrist_3_lower_limit:=${-pi}    wrist_3_upper_limit:=${pi}
     **transmission_hw_interface:=hardware_interface/PositionJointInterface"**
	 >

before :
	<xacro:ur_arm_transmission prefix="${prefix}" />
after :
	<xacro:ur_arm_transmission prefix="${prefix}" **hw_interface="${transmission_hw_interface}"** />

but i don't know (in this ur5.urdf.xacro) whether we should change **shoulder_pan_lower_limit** and **shoulder_pan_upper_limit** into **-2*pi** and **2*pi** or not. Does anyone know?

Brant714, 2019-11-05:
> i try it again with the another version of universal robot,but the error still exist,just change the path's name.following is the error:
> **Invalid parameter "transmission_hw_interface" None None**
> when instantiating macro: ur5_robot (/home/hourui/catkin_ws/src/universal_robot/ur_description/urdf/ur5.urdf.xacro)
> in file: /home/hourui/catkin_ws/src/universal_robot/ur_description/urdf/ur5_joint_limited_robot.urdf.xacro
> ### while processing /home/hourui/catkin_ws/src/universal_robot/ur_description/launch/ur5_upload.launch:
> 
> Invalid tag: Cannot load command parameter [robot_description]: command [/opt/ros/kinetic/lib/xacro/xacro --inorder '/home/hourui/catkin_ws/src/universal_robot/ur_description/urdf/ur5_joint_limited_robot.urdf.xacro' transmission_hw_interface:=hardware_interface/PositionJointInterface] returned with code [2].
> ### Param xml is
> 
> The traceback for the exception was written to the log file
> 
> can i add your wechat?i really want fix it with your help!

I've been stuck with this problem for two days, right? Can you tell me how you solved it?

cyu-craftnetics, 2020-02-29:
I'm also experiencing this problem. Anyone have any updates on resolving this?

Priyanka328, 2020-03-18:
Hi,I am facing also the same problem .do someone know what to do?

niuxiansen2, 2020-03-28:
> 嗨，我也面临着同样的问题。有人知道该怎么办吗？

 This repo was based on kinetic version，you might use kinetic-devel version of Universal Robot, but. This might be a possible reason.

cyu-craftnetics, 2020-04-01:
Hi @Priyanka328,

I was able to get this working. I added 

> transmission_hw_interface:=hardware_interface/PositionJointInterface

to the <xacro:macro name="ur5_robot" tag in the ur5.urdf.xacro file. It should look like this below:
[image]

I also added

> hw_interface="${transmission_hw_interface}

to the "camera_link_optical" tag in the ur5.urdf.xacro file. It should look like this

[image]

BigRays, 2020-05-20:
> 问题<1>～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～
> Invalid parameter "transmission_hw_interface" None None
> when instantiating macro: ur5_robot [...]

very nice!!!  Thank you very much~

learningxiaobai, 2021-01-07:
hi，i want to know if someone work it out using melodic and Ubuntu18.04，
i will appreciate if someone help me out

learningxiaobai, 2021-01-07:
> hello,lihuang,when i replace these two files with customized files. Under directory ur_ws/src/ur5_ROS-Gazebo/src/ur_description/, copy common.gazebo.xacro and ur5.urdf.xacro to ur_ws/src/universal_robot/ur_description/urdf/.then i roslunch ur5_notebook initialize.launch,there ia an error makes can not start the gazebo,i am so confused,would you help me,folling is the detail about error:
> 
> ###(ERROR) Invalid parameter "transmission_hw_interface" None ### None
> when instantiating macro: ur5_robot (/home/hourui/catkin_ws/src/universal_robot-kinetic-devel/universal_robot-kinetic-devel/ur_description/urdf/ur5.urdf.xacro)
> in file: /home/hourui/catkin_ws/src/universal_robot-kinetic-devel/universal_robot-kinetic-devel/ur_description/urdf/ur5_joint_limited_robot.urdf.xacro
> 
> ### {ERROR}while processing /home/hourui/catkin_ws/src/universal_robot-kinetic-devel/universal_robot-kinetic-devel/ur_description/launch/ur5_upload.launch:
> Invalid tag: Cannot load command parameter [robot_description]: command [/opt/ros/kinetic/lib/xacro/xacro --inorder '/home/hourui/catkin_ws/src/universal_robot-kinetic-devel/universal_robot-kinetic-devel/ur_description/urdf/ur5_joint_limited_robot.urdf.xacro' transmission_hw_interface:=hardware_interface/PositionJointInterface] returned with code [2].
> 
> Param xml is
> The traceback for the exception was written to the log file
> 
> waiting for your help,grateful about you!

hello ，you are using melodic or not

learningxiaobai, 2021-01-07:
> > 嗨，我也面临着同样的问题。有人知道该怎么办吗？
> 
> This repo was based on kinetic version，you might use kinetic-devel version of Universal Robot, but. This might be a possible reason.

you mean ,melodic can't do that?

learningxiaobai, 2021-01-07:
hello， i use melodic，but error occurs。
RLException: while processing /home/xiong/ur5_ws/src/universal_robot/ur_description/launch/ur5_upload.launch:
Invalid <param> tag: Cannot load command parameter [robot_description]: command [['/opt/ros/melodic/lib/xacro/xacro', '--inorder', '/home/xiong/ur5_ws/src/universal_robot/ur_description/urdf/ur5_joint_limited_robot.urdf.xacro', 'transmission_hw_interface:=hardware_interface/PositionJointInterface']] returned with code [2]. 

Param xml is <param command="$(find xacro)/xacro --inorder '$(find ur_description)/urdf/ur5_joint_limited_robot.urdf.xacro' transmission_hw_interface:=$(arg transmission_hw_interface)" if="$(arg limited)" name="robot_description"/>
The traceback for the exception was written to the log file

i hope someone can help me，please

Rudy112, 2021-11-07:
> hello， i use melodic，but error occurs。 RLException: while processing /home/xiong/ur5_ws/src/universal_robot/ur_description/launch/ur5_upload.launch: Invalid tag: Cannot load command parameter [robot_description]: command [['/opt/ros/melodic/lib/xacro/xacro', '--inorder', '/home/xiong/ur5_ws/src/universal_robot/ur_description/urdf/ur5_joint_limited_robot.urdf.xacro', 'transmission_hw_interface:=hardware_interface/PositionJointInterface']] returned with code [2].
> 
> Param xml is The traceback for the exception was written to the log file
> 
> i hope someone can help me，please

I also use ros-melodic, facing the same issue. Did you solve that?

MikasaIL, 2024-04-20:
[bin-10] process has finished cleanly
log file: /home/zyb/.ros/log/46d5c0e4-fcb5-11ee-9095-90de80b58653/bin-10*.log
gzserver: /build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreRenderSystem.cpp:546: virtual void Ogre::RenderSystem::setDepthBufferFor(Ogre::RenderTarget*): Assertion `bAttached && "A new DepthBuffer for a RenderTarget was created, but after creation" "it says it's incompatible with that RT"' failed.
[spawn_conveyor_belt-9] process has finished cleanly
log file: /home/zyb/.ros/log/46d5c0e4-fcb5-11ee-9095-90de80b58653/spawn_conveyor_belt-9*.log
[joint_state_controller_spawner-6] process has finished cleanly
log file: /home/zyb/.ros/log/46d5c0e4-fcb5-11ee-9095-90de80b58653/joint_state_controller_spawner-6*.log
[ INFO] [1713615961.814385644, 0.586000000]: Ready to take commands for planning group manipulator.
[ INFO] [1713615961.814519800, 0.586000000]: Replanning: yes
[ INFO] [1713615962.817016162, 0.586000000]: Didn't received robot state (joint angles) with recent timestamp within 1 seconds.
Check clock synchronization if your are running ROS across multiple machines!
[ERROR] [1713615962.817059938, 0.586000000]: Failed to fetch current robot state
[ INFO] [1713615963.817984433, 0.586000000]: Didn't received robot state (joint angles) with recent timestamp within 1 seconds.
Check clock synchronization if your are running ROS across multiple machines!
[ERROR] [1713615963.818235155, 0.586000000]: Failed to fetch current robot state
Traceback (most recent call last):
  File "/home/zyb/ur_ws/src/ur5_ROS-Gazebo/ur5_mp.py", line 374, in <module>
    mp=ur5_mp()
  File "/home/zyb/ur_ws/src/ur5_ROS-Gazebo/ur5_mp.py", line 125, in __init__
    self.default_joint_states[0] = -1.57691
IndexError: list assignment index out of range

sorry,I got this problem,could anyone help me ?thank u very much!!