lihuang3/ur5_ROS-Gazebo #14: How can I adjust the speed of the red  box and the arm?

wfc199821, 2019-05-28:
Hi,lihuang3,I want to know adjust the speed of the red  box and the arm.I have tried modify the configuration file(joint_limits.yaml),the robot arm can grab the box,but then it fell off.I hope the box and arm can move fast,but now I don't know how to make it.Hope then you can give some suggestions,thanks.

lihuang3, 2019-05-28:
@wfc199821 As for the arm, you need to change the speed limit of the UR5, which is out of the scope of this project. @Haoran-Zhao Can you help him/her with the box speed?

Haoran-Zhao, 2019-05-28:
@wfc199821 you can change the speed of the red box in the file [blocks_spawner.cpp](https://github.com/lihuang3/ur5_ROS-Gazebo/blob/master/blocks_spawner.cpp).  the movement of the red box is generated by apply body wrench. 
`      apply_wrench_req.wrench.force.x = -5.1;
      apply_wrench_req.wrench.force.y = 0.0;
      apply_wrench_req.wrench.force.z = 0.0;
      apply_wrench_req.start_time = time_temp;
      apply_wrench_req.duration = duration_temp;
      apply_wrench_req.reference_frame = "world";`