lihuang3/ur5_ROS-Gazebo #15: How can i move my arm and turn the camera when all models are present

Just17, 2019-05-29:
Hi,lihuang3,when i run **roslaunch ur5_notebook initialize.launch** i can see the whole models,but my arm can't move ,and the camera window can appear but there is no image.Also there ia an error :  

 **[ur5_mp-15] process has died [pid 3924, exit code 1, cmd /home/just17/catkin_ws/src/ur5_ROS-Gazebo-master/ur5_mp.py /follow_joint_trajectory:=/arm_controller/follow_joint_trajectory __name:=ur5_mp __log:=/home/just17/.ros/log/3f6e8a5e-8206-11e9-a337-000c2900ddbe/ur5_mp-15.log].
log file: /home/just17/.ros/log/3f6e8a5e-8206-11e9-a337-000c2900ddbe/ur5_mp-15*.log**

I am very intersted about it and try many times . how can i deal with this error and move my arm ? Looking forward to your reply.Thank you very much

lihuang3, 2019-05-29:
@Just17 Are you able to manipulate UR5 in rviz with moveit?

Just17, 2019-05-30:
@lihuang3 Thank you for your reply.i can do it, it is working fine after run **python ur5_mp.py**in the new terminal .but i saw you said only need this **roslaunch ur5_notebook initialize.launch**  Why can't I just run this command successfully?

I would also like to learn more about multi-object recognition, and Is it possible to change the vacuum grip to another grip?Looking forward to your reply.Thank you very much

lihuang3, 2019-05-31:
@Just17 Looks like that the scenario `[ur5_mp-15] process has died` might refer to some issues with ROS/Gazebo.  I'm not quite sure about this case. Reinstalling some related packages may help, or maybe simply kill ROS and restart?

Sure you can change the gripper, and the vacuum gripper is just a simple example.

Just17, 2019-06-03:
@lihuang3 Thank you very much. I'll keep trying.

zhouaiChaichai, 2019-06-10:
I meet the same problem ,there is anybody solved it? tanks a lot

learningxiaobai, 2021-04-18:
> @lihuang3 Thank you for your reply.i can do it, it is working fine after run **python ur5_mp.py**in the new terminal .but i saw you said only need this **roslaunch ur5_notebook initialize.launch** [...]

hello ,after run **python ur5_mp.py**in the new terminal,the arm still can't move,why