lihuang3/ur5_ROS-Gazebo #19: why i cann't see the camera window when i roslaunch

Just17, 2019-07-12:
The moveit software was deleted by mistake due to the installation of other software,After I reinstalled it, I found that gazebo could display scene objects, but the camera window did not appear and did not grab them. What's the matter? Did I delete the software that opened the window by mistake?

Looking forward to your reply

lihuang3, 2019-07-12:
Did you see a green camera module on the end effector? Please refer to this [issue] ('https://github.com/lihuang3/ur5_ROS-Gazebo/issues/3#issuecomment-420861652')

Just17, 2019-07-13:
@lihuang3 Thank you for your reply.I have all the modules,But the camera window doesn't open.I used to be able to run and grab well, but after I installed some other software, the situation appeared when I opened this one. So I would like to ask whether the camera window module is associated with other ones, and whether some of them were deleted by mistake?

Looking forward to your reply