lihuang3/ur5_ROS-Gazebo #20: Using Two Robots for the Pick and Place Task

Murtaza786-oss, 2019-09-16:
Hello, 
The code seems to work perfectly but now I am planning to use two UR5 instead one for the same task.
I can successfully launch two robots but then it doesn't do anything.

Here is my launch file for launching the two robots 

```
   <?xml version="1.0"?>
<launch>
  <param name="red_box_path" type="str" value="$(find ur5_notebook)/urdf/red_box.urdf"/>
  <param name="/use_sim_time" value="true" />
  <arg name="robot_name"/>
  <arg name="init_pose"/>

  <arg name="limited" default="true"/>
  <arg name="paused" default="false"/>
  <arg name="gui" default="true"/>
  <arg name="debug" default="false" />
  <arg name="sim" default="true" />
  <arg name="transmission_hw_interface" default="hardware_interface/PositionJointInterface" />
  

  <!-- startup simulated world -->
  <include file="$(find gazebo_ros)/launch/empty_world.launch">
    <arg name="world_name" default="worlds/empty.world"/>
    <arg name="paused" value="$(arg paused)"/>
    <arg name="gui" value="$(arg gui)"/>
  </include>

  <!-- spawn the conveyor_belt in gazebo -->
  <node name="spawn_conveyor_belt" pkg="gazebo_ros" type="spawn_model" args="-file $(find ur5_notebook)/urdf/conveyor_belt.urdf -urdf -model conveyor_belt" />
  <!-- spawn the conveyor_belt in gazebo -->
  <node name="bin" pkg="gazebo_ros" type="spawn_model" args="-file $(find ur5_notebook)/urdf/bin.urdf -urdf -model bin -y 0.8 -x -0.5 -z 0.05" />
  <!-- the red blocks spawner node -->
  <node name="blocks_spawner" pkg="ur5_notebook" type="blocks_spawner" output="screen" />

  <!-- the cylinder poses publisher node -->
  <node name="blocks_poses_publisher" pkg="ur5_notebook" type="blocks_poses_publisher" output="screen" />

  <!-- spwan ur5 -->
  <!-- send robot urdf to param server -->
  <group ns="robot1">
    <param name="tf_prefix" value="robot1_tf" />
        <include file="$(find ur_gazebo)/launch/ur5.launch">
                <arg name="init_pose" value="-z 0.2 -y 0.7"/>
                <arg name="robot_name" value="robot1"/>
        </include>

        <include file="$(find ur5_moveit_config)/launch/move_group.launch">
          <arg name="limited" default="$(arg limited)"/>
          <arg name="debug" default="$(arg debug)" />
        </include>
        
        <node name="ur5_vision" pkg="ur5_notebook" type="ur5_vision.py" output="screen" />
        <node name="ur5_mp" pkg="ur5_notebook" type="ur5_mp.py" output="screen" />
        <node name="ur5_gripper" pkg="ur5_notebook" type="ur5_gripper.py" output="screen"/>

        <remap if="$(arg sim)" from="follow_joint_trajectory" to="arm_controller/follow_joint_trajectory"/>

        <!-- Remap follow_joint_trajectory -->

        <!-- Launch moveit -->

          <!-- the cylinder poses publisher node -->

  </group>

  <group ns="robot2">
    <param name="tf_prefix" value="robot2_tf" />
        <include file="$(find ur_gazebo)/launch/ur5.launch">
                <arg name="init_pose" value="-z 0.2 -y -0.7"/>
                <arg name="robot_name" value="robot2"/>
        </include>

        <include file="$(find ur5_moveit_config)/launch/move_group.launch">
          <arg name="limited" default="$(arg limited)"/>
          <arg name="debug" default="$(arg debug)" />
        </include>
        
        <node name="ur5_vision" pkg="ur5_notebook" type="ur5_vision.py" output="screen" />
        <node name="ur5_mp" pkg="ur5_notebook" type="ur5_mp.py" output="screen" />
        <node name="ur5_gripper" pkg="ur5_notebook" type="ur5_gripper.py" output="screen"/>

        <remap if="$(arg sim)" from="follow_joint_trajectory" to="arm_controller/follow_joint_trajectory"/>

        <!-- Remap follow_joint_trajectory -->

        <!-- Launch moveit -->

          <!-- the cylinder poses publisher node -->

  </group>

  <!-- for ros control to be used with scara robot -->
  <!--   <param name="/scara_robot_left/robot_description" textfile="$(find two_scara_collaboration)/urdf/scara_robot_left.urdf" />-->

<!-- spawn the red_box in gazebo -->
<!-- node name="spawn_red_box" pkg="gazebo_ros" type="spawn_model" args="-file $(find ur5_notebook)/urdf/red_box.urdf -urdf -model red_box"/ -->

</launch>
```

when I launch the above file the robots are launched but there is no execution of the 'ur5_mp.py'
I get the following error

```
Traceback (most recent call last):
  File "/home/murtaza/catkin_ws/src/ur5_notebook/ur5_mp.py", line 378, in <module>
    mp=ur5_mp()
  File "/home/murtaza/catkin_ws/src/ur5_notebook/ur5_mp.py", line 66, in __init__
    self.arm = moveit_commander.MoveGroupCommander(group_name)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/moveit_commander/move_group.py", line 52, in __init__
    self._g = _moveit_move_group_interface.MoveGroupInterface(name, robot_description, ns)
RuntimeError: Unable to connect to move_group action server 'move_group' within allotted time (5s)
[INFO] [1568641657.813389, 0.000000]: Stopping the robot
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/core.py", line 572, in signal_shutdown
    h()
  File "/home/murtaza/catkin_ws/src/ur5_notebook/ur5_mp.py", line 157, in cleanup
    self.arm.stop()
AttributeError: ur5_mp instance has no attribute 'arm'
[robot1/ur5_mp-15] process has died [pid 35212, exit code 1, cmd /home/murtaza/catkin_ws/src/ur5_notebook/ur5_mp.py __name:=ur5_mp __log:=/home/murtaza/.ros/log/c6d316e8-d86b-11e9-a0db-000c29544a54/robot1-ur5_mp-15.log].
log file: /home/murtaza/.ros/log/c6d316e8-d86b-11e9-a0db-000c29544a54/robot1-ur5_mp-15*.log
[robot2/ur5_mp-25] process has died [pid 35239, exit code 1, cmd /home/murtaza/catkin_ws/src/ur5_notebook/ur5_mp.py __name:=ur5_mp __log:=/home/murtaza/.ros/log/c6d316e8-d86b-11e9-a0db-000c29544a54/robot2-ur5_mp-25.log].
log file: /home/murtaza/.ros/log/c6d316e8-d86b-11e9-a0db-000c29544a54/robot2-ur5_mp-25*.log
```
Here is the snippet of the result after execution

[image]

lihuang3, 2019-09-18:
@Murtaza786-oss I remember we had the same issue a year ago, but we didn't figure it out at that time.....

Murtaza786-oss, 2019-09-27:
Hello I have successfully solved controlling two robots together for the pick and place task. I have shared the image in the below link. I still have some small issue with gripping action it would be glad if you can help me out 
https://drive.google.com/file/d/1-2-torZESz9L5LwnuYmeqBtqsBQGFyWg/view?usp=sharing
https://drive.google.com/open?id=1-0OOvLoSEtMgCnLFvc9zETEE0UTqCroC

Murtaza786-oss, 2019-09-30:
Hi I have successfully solved the pick and place task by coordinating two UR5. You can find the working video here on youtube

[https://youtu.be/ZS8rTd0JSuU]

Murtaza786-oss, 2019-09-30:
Hi I have successfully solved the pick and place task by coordinating two UR5. You can find the working video here on youtube

[https://youtu.be/ZS8rTd0JSuU](url)

lihuang3, 2019-10-01:
@Murtaza786-oss Good job!

63445538, 2019-11-30:
> Hi I have successfully solved the pick and place task by coordinating two UR5. You can find the working video here on youtube
> 
> https://youtu.be/ZS8rTd0JSuU

Hi, could you share your implementation about the two arm simulation? I am stick in it. Thank you.

closed 2020-04-19 by lihuang3

ar-i-enterprise, 2020-08-01:
> Hi I have successfully solved the pick and place task by coordinating two UR5. You can find the working video here on youtube
> 
> https://youtu.be/ZS8rTd0JSuU

Hallo Muztafa, How did you solve the problem? I still facing the same problem. 

`Traceback (most recent call last):
  File "/home/murtaza/catkin_ws/src/ur5_notebook/ur5_mp.py", line 378, in <module>
    mp=ur5_mp()
  File "/home/murtaza/catkin_ws/src/ur5_notebook/ur5_mp.py", line 66, in __init__
    self.arm = moveit_commander.MoveGroupCommander(group_name)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/moveit_commander/move_group.py", line 52, in __init__
    self._g = _moveit_move_group_interface.MoveGroupInterface(name, robot_description, ns)
RuntimeError: Unable to connect to move_group action server 'move_group' within allotted time (5s)
[INFO] [1568641657.813389, 0.000000]: Stopping the robot
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/core.py", line 572, in signal_shutdown
    h()
  File "/home/murtaza/catkin_ws/src/ur5_notebook/ur5_mp.py", line 157, in cleanup
    self.arm.stop()
AttributeError: ur5_mp instance has no attribute 'arm'
[robot1/ur5_mp-15] process has died [pid 35212, exit code 1, cmd /home/murtaza/catkin_ws/src/ur5_notebook/ur5_mp.py __name:=ur5_mp __log:=/home/murtaza/.ros/log/c6d316e8-d86b-11e9-a0db-000c29544a54/robot1-ur5_mp-15.log].
log file: /home/murtaza/.ros/log/c6d316e8-d86b-11e9-a0db-000c29544a54/robot1-ur5_mp-15*.log
[robot2/ur5_mp-25] process has died [pid 35239, exit code 1, cmd /home/murtaza/catkin_ws/src/ur5_notebook/ur5_mp.py __name:=ur5_mp __log:=/home/murtaza/.ros/log/c6d316e8-d86b-11e9-a0db-000c29544a54/robot2-ur5_mp-25.log].
log file: /home/murtaza/.ros/log/c6d316e8-d86b-11e9-a0db-000c29544a54/robot2-ur5_mp-25*.log`

ar-i-enterprise, 2020-08-01:
> @Murtaza786-oss I remember we had the same issue a year ago, but we didn't figure it out at that time.....

I am facing the problem. How did you solve this problem?