lihuang3/ur5_ROS-Gazebo #22: Conversion of pixel coordinate system and manipulator coordinate system

richardsanqi, 2020-03-31:
hello，lihuang，how do you just use  camera to get the real point to make the robot move ？i know it needs calibration.

lihuang3, 2020-03-31:
@richardsanqi Those are just coordinates in pixel space, not in real 3D space.

richardsanqi, 2020-04-01:
> @richardsanqi Those are just coordinates in pixel space, not in real 3D space.

but how you make the robot just to catch the red box.I learn your code just two days,i can't find the tf transofm  just like this code ` this->tf_camera_to_robot.lookupTransform("/base_link", "/camera_link", ros::Time(0), (this->camera_to_robot_));`

lihuang3, 2020-04-01:
@richardsanqi Objects are assumed to be on the flat plane, so their height is fixed. We only need to take care of the horizontal plane locations.

richardsanqi, 2020-04-05:
> @richardsanqi假定对象位于平面上，因此它们的高度是固定的。我们只需要照顾水平面的位置。
thank you,but i have someNot sure. in ur5_mp.py file the code `wpose.position.y=self.erroy_y*0.04/105` just mean change to the robot end link coordinates ? and how did you get the 0.04/105, did you use the red box width and length to calculate the number，so as 0.05/105.the last question is how did  you get the number "195" `self.error_y=self.cy-(h/2+195)` in ur5_vision.py

closed 2020-04-19 by lihuang3