lihuang3/ur5_ROS-Gazebo #24: The robot arm doesn't pick the boxes

Egehae, 2020-06-09:
Hi, lihuang. I'm new to ROS and interested in your project,
because I wanna simulate picking process with vacuum gripper.

But when I launch `initialize.launch`, the robot arm doesn't pick the red boxes though it detects them.
It just goes to the placing position with empty hand and stays still.

As I'm using melodic, I thought there could be some errors but can you help me with it?

[image]
[image]

lihuang3, 2020-06-09:
I had similar issues before. You can just try restarting the program/ROS-gazebo. If it doesn't work, it might be the ROS version problem as it was only tested on Kinetic.

Egehae, 2020-06-09:
I solved it! The problem was in `ur5_mp.py`. I don't know whether it's because of the version difference, but I fixed the value of `reference_frame = "/base_link"` to `"base_link"` and now the simulation works well. Thanks for this great work.

closed 2020-06-09 by Egehae

Motion77, 2022-04-29:
> I solved it! The problem was in `ur5_mp.py`. I don't know whether it's because of the version difference, but I fixed the value of `reference_frame = "/base_link"` to `"base_link"` and now the simulation works well. Thanks for this great work.

Hello, can you directly change the code from "/base_link" to "base_link"?

Motion77, 2022-04-30:
> I solved it! The problem was in `ur5_mp.py`. I don't know whether it's because of the version difference, but I fixed the value of `reference_frame = "/base_link"` to `"base_link"` and now the simulation works well. Thanks for this great work.

hello，Can share your open source project code？