lihuang3/ur5_ROS-Gazebo #26: The arm can't move and the camear

aubiao, 2020-07-23:


aubiao, 2020-07-23:
the terminal print warn when the red box cross the camera
--------------------------------------------------------------------------
[ WARN] [1595472688.063235819, 117.205000000]: The complete state of the robot is not yet known.  Missing gripper_joint, gripper_joint1, gripper_joint2, gripper_joint3, gripper_joint4, gripper_joint5, gripper_joint6, gripper_joint7, gripper_joint8
[ WARN] [1595472695.682415864, 118.245000000]: The complete state of the robot is not yet known.  Missing gripper_joint, gripper_joint1, gripper_joint2, gripper_joint3, gripper_joint4, gripper_joint5, gripper_joint6, gripper_joint7, gripper_joint8
[ WARN] [1595472703.519119242, 119.245000000]: The complete state of the robot is not yet known.  Missing gripper_joint, gripper_joint1, gripper_joint2, gripper_joint3, gripper_joint4, gripper_joint5, gripper_joint6, gripper_joint7, gripper_joint8
-----------------------------------------------------

also, some error
--------------------------------------------------------
[ERROR] [1595471981.780210349, 0.130000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/shoulder_pan_joint
[ERROR] [1595471981.784470315, 0.130000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/shoulder_lift_joint
[ERROR] [1595471981.788446378, 0.130000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/elbow_joint
[ERROR] [1595471981.794961431, 0.130000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/wrist_1_joint
--------------------------------------------------------------
I can't solve the problem,please help me,thanks.

aubiao, 2020-07-23:
[image]

learningxiaobai, 2021-01-07:
> ## the terminal print warn when the red box cross the camera
> ## [ WARN] [1595472688.063235819, 117.205000000]: The complete state of the robot is not yet known. Missing gripper_joint, [...]

have you solved？i have same questions

zcz-vvv, 2021-02-21:
@learningxiaobai  i have same questions,have you solved?please help me,thank you

zcz-vvv, 2021-02-21:
@aubiao  i have same questions,have you solved?please help me,thank you

learningxiaobai, 2021-04-18:
> @learningxiaobai i have same questions,have you solved?please help me,thank you

sorry，i have't ,have you solved?

Jayce-Jayce, 2021-05-30:
have you guys solved? i met the same thing.

weakenleg, 2021-07-10:
have you solved? i met the same thing.

hxn8439, 2021-09-04:
I have the same problem too. The robot arm suppose to move after executing
roslaunch ur5_notebook initialize.launch

So, I also execute python testvisionplanner.py and testmove2.py but endup with this error return.

[ERROR] [1630771695.765029]: Unable to immediately register with master node [http://localhost:11311]: master may not be running yet. Will keep trying.

so the error is happening at the master node communication.

onehuan, 2021-11-25:
have you guys solved? i met the same thing.

onehuan, 2021-11-25:
> 

have you guys solved? i met the same thing.

onehuan, 2021-11-25:
> I have the same problem too. The robot arm suppose to move after executing roslaunch ur5_notebook initialize.launch
>
> So, I also execute python testvisionplanner.py and testmove2.py but endup with [...]

have you guys solved? i met the same thing.

onehuan, 2021-11-25:
> have you solved? i met the same thing.

have you guys solved? i met the same thing.

hxn8439, 2021-12-26:
Yes, it is resolved, it turns out that I have to converted to melodic distribution. You can find the repository with a melodic distribution at https://github.com/hxn8439/ur5_ROS-Gazebo-Melodic. I listed a github user, lihuang3, as the main contributor of the work.

onehuan, 2021-12-26:
Thank you very much, and I will try it agian.

hwan30, 2022-04-07:
Is using melodic the only fix to this issue?
I have used both kinetic and noetic and same issue persists.

Motion77, 2022-04-29:
> have you guys solved? i met the same thing.
have you guys solved? i met the same thing.

Motion77, 2022-04-29:
> have you guys solved? i met the same thing.

Have you guys solved? I met the same thing.

Motion77, 2022-04-29:
> Thank you very much, and I will try it agian.
> […](#)

Have you guys solved? I met the same thing.

onehuan, 2022-04-29:
> > Thank you very much, and I will try it agian.
> > […](#)
> 
> Have you guys solved? I met the same thing.

sorry, i didn't solve the problem

Motion77, 2022-04-30:
> Yes, it is resolved, it turns out that I have to converted to melodic distribution. You can find the repository with a melodic distribution at https://github.com/hxn8439/ur5_ROS-Gazebo-Melodic. I listed a github user, lihuang3, as the main contributor of the work.

Hello, when I run the melodic version, the red object cannot move, have you encountered it?

Motion77, 2022-04-30:
> Is using melodic the only fix to this issue? I have used both kinetic and noetic and same issue persists.

have you guys solved? i met the same thing.

Motion77, 2022-05-02:
> Yes, it is resolved, it turns out that I have to converted to melodic distribution. You can find the repository with a melodic distribution at https://github.com/hxn8439/ur5_ROS-Gazebo-Melodic. I listed a github user, lihuang3, as the main contributor of the work.
Hello, using the melodic version, is there anything to modify?

Motion77, 2022-05-02:
> Yes, it is resolved, it turns out that I have to converted to melodic distribution. You can find the repository with a melodic distribution at https://github.com/hxn8439/ur5_ROS-Gazebo-Melodic. I listed a github user, lihuang3, as the main contributor of the work.
can you give me your contact information？

2000222, 2022-08-31:
I run the melodic version code with ros noetic, but the robot can not move either, even with the following issues: 

[ERROR] [1661952199.436096, 3.621000]: bad callback: <bound method ur5_mp.tracking_callback of <__main__.ur5_mp object at 0x7f1a6be80ee0>>
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/sarah/ur5_ws/src/ur5_ROS-Gazebo-Melodic/ur5_mp.py", line 167, in tracking_callback
    if len(self.pointx)>9:
AttributeError: 'ur5_mp' object has no attribute 'pointx'