lihuang3/ur5_ROS-Gazebo #30: melodic

learningxiaobai, 2021-01-07:
hi，is there someone work it out using melodic？

learningxiaobai, 2021-01-07:
[ERROR] [1610054610.045944855]: The kinematics plugin (manipulator) failed to load. Error: Failed to load library /opt/ros/melodic/lib//libmoveit_kdl_kinematics_plugin.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libmoveit_kinematics_base.so.1.0.7: cannot open shared object file: No such file or directory)
[ERROR] [1610054610.046215613]: Kinematics solver could not be instantiated for joint group manipulator.

can  someone help me out？

ashishagg70, 2021-06-24:
Yes, I could run with melodic.
https://github.com/lihuang3/ur5_ROS-Gazebo/issues/32#issuecomment-867516863

Motion77, 2022-04-30:
Melodic version, my robotic arm can't pick up red objects, have you ever encountered it?

TD-zane, 2022-05-05:
> Melodic version, my robotic arm can't pick up red objects, have you ever encountered it?

Hello,did you solve the problem running on Melodic?