lihuang3/ur5_ROS-Gazebo #32: Invalid parameter "transmission_hw_interface"

mingmou123, 2021-04-29:
is there someone meet the same problem?
I guess it's because universal-robot official add transmission_hw_interface which cause the ur5.urdf.xacro is not consistant with the u5_joint_limited_robot.urdf.xacro.
when i  delete the hw_interface, i met another problem "XML parsing error: mismatched tag: line 66, column 4

ashishagg70, 2021-06-24:
can you attach your ur5.urdf.xacro file.

Motion77, 2022-04-30:
> can you attach your ur5.urdf.xacro file. or just see it should be like following:
> 
> `<xacro:macro name="ur5_robot" params="prefix joint_limited shoulder_pan_lower_limit:=${-2_pi} shoulder_pan_upper_limit:=${2_pi} shoulder_lift_lower_limit:=${-pi} shoulder_lift_upper_limit:=${pi} elbow_joint_lower_limit:=${-pi} elbow_joint_upper_limit:=${pi} wrist_1_lower_limit:=${-pi} wrist_1_upper_limit:=${pi} wrist_2_lower_limit:=${-pi} wrist_2_upper_limit:=${pi} wrist_3_lower_limit:=${-pi} wrist_3_upper_limit:=${pi} transmission_hw_interface:=hardware_interface/PositionJointInterface">
> 
> . . . . <xacro:ur_arm_transmission prefix="${prefix}" hw_interface="hardware_interface/PositionJointInterface"/>`
> 
> as suggested in [#13 (comment)](https://github.com/lihuang3/ur5_ROS-Gazebo/issues/13#issuecomment-526238816)

nice